PiPHP/GPIO #3: Set direction in constructor?

colinodell, 2016-05-12:
Many other GPIO libraries require you to set the direction at the same time the pin is declared for use - either via the constructor or some other initialization function.  This makes sense, because you're probably never going to change the direction of a pin during execution without rewiring your project.

Would it make sense to do the same thing with the `Pin` class?  Basically change its constructor to include a `$direction` parameter?  Or alternatively create separate `InputPin` and `OutputPin` classes, where the `setValue()` method is only available on the latter?

I'd be happy to submit a PR for either change if you're interested.

AndrewCarterUK, 2016-05-12:
@colinodell - I like this idea, and I think there is an overlap here with the other issue that you created (https://github.com/PiPHP/GPIO/issues/4).

At the moment the library is a very thin wrapper over the sysfs GPIO driver, but I think it can become a lot easier to use with some of the changes you mention.

I'd gladly accept a PR on this if you're willing :)

I can see scenarios where you might want a pin to change (if you created plugs for the GPIO and had different modules that might be plugged in or not) - but it wouldn't stop you from making those changes. The calling code could easily just call the factory again asking for a new pin.

AndrewCarterUK, 2016-05-12:
What do you think about having:

```php
PinFactoryInterface::getInputPin(): InputPinInterface;
PinFactoryInterface::getOutputPin(): OutputPinInterface;
```

Then we could put the suitable methods onto each interface.

colinodell, 2016-05-12:
Yeah that would be perfect!

closed 2016-05-17 by AndrewCarterUK