PiPHP/GPIO #4: Pin::getEdge/setEdge

colinodell, 2016-05-12:
Edge detection is primarily used for interrupt detection - sometimes we only care about interrupts LOW -> HIGH for example.  However, it looks like this is being set via the `Pin` class (and can be done for output pins, which isn't useful).  Would it make more sense to set this on the interrupt watcher instead?

AndrewCarterUK, 2016-05-12:
Let me give this some thought!

Trying to decide if this is best as a parameter to `InterruptWatcher::register()`, a parameter to the input pin factory or a method on the input pin.

AndrewCarterUK, 2016-05-12:
I think my main concern about having this is an `InterruptWatcher::register()` parameter or factory parameter is that it implies it wont change.

I'm leaning towards methods on an `InputPin` class.

colinodell, 2016-05-12:
I feel it's more logical to define the edge you're interested in detecting at the same time you're registering the callback to handle occurrences of that edge.

For example, let's say I want a callback to fire whenever a button is pressed.  I'd want to listen for any FALLING (or RISING) events and do some action.  However, hooking that same callback up to a pin set to BOTH means the callback would fire twice.  If I forget to set the pin's edge beforehand, or if it's modified at some other point in time, my callback isn't going to work as expected.

For Arduino, setting the edge is done when registering the interrupt callback: https://www.arduino.cc/en/Reference/AttachInterrupt

RPi.GPIO does the same thing with their `wait_for_edge()` function: https://sourceforge.net/p/raspberry-gpio-python/wiki/Inputs/