PiPHP/GPIO #17: Permissions are not set, pelase provide example

Erfeniuses, 2017-03-29:
Hi There.

I am sorry to bring this up, but could the permissions settings also be addressed here, because this does not work without them.

When user starts this from the pure clean jessie image and installs apache2 with php5, GPIO permissions are not right

Thank you for your considerations.

AndrewCarterUK, 2017-03-29:
Hello,

You may find this document useful: https://github.com/ronanguilloux/php-gpio#understanding-io-permissions

Erfeniuses, 2017-03-29:
Thanx, but could you write proper guide here as well... php files are usually at /var/www/html/ index.php e.g.

AndrewCarterUK, 2017-03-29:
Hi,

There's unlikely to be a proper guide here unless someone else writes one and submits a PR. I'm not actively developing this project currently (I've got quite a lot on) but I'll try and keep up to date with bugs.

I'd install this library just as you would any composer library (into the same directory as your web project - probably /var/www/html).

Erfeniuses, 2017-04-02:
ok. thanx. still need those permission settings, none of the net instructions work. :(

closed 2020-08-08 by svpernova09