PiPHP/GPIO #18: How to use PWM ?

alcalyn, 2017-09-26:
I didn't found any PWM support in this library, how do you achieve this ?

erikaheidi, 2018-06-23:
+1 to that question! I've been wondering how to use PWM with this library.

AndrewCarterUK, 2018-06-23:
This library uses the `sysfs` driver for controlling the GPIO pins, and this driver doesn't support PWM.

I'd suggest using a library such as [phpi](https://github.com/calcinai/phpi) as it uses different kernel interfaces to control the pins and can thus support PWM.

Sorry!

AndrewCarterUK, 2018-06-23:
It does seem like [there is a similar approach that uses a driver similar to the sysfs driver](http://blog.oddbit.com/2017/09/26/some-notes-on-pwm-on-the-raspberry-pi/).

If anyone reading this wants to have a crack at `PiPHP/PWM` - I'd be more than happy to support.

AndrewCarterUK, 2018-06-23:
I've set up a [PiPHP/PWM](https://github.com/piphp/pwm) repo that people can fork from if they want to play around with this as a project.

I've stripped it down to a single pin class, and you can see most of the important changes [in this diff](https://github.com/PiPHP/PWM/commit/5e766183988b3042dcaa5495a609e130714c51fc#diff-3c80025ce9c0b3e1189ebdda7ef5d99e).

svpernova09, 2020-08-08:
Seems like this has been resolved via https://github.com/piphp/pwm.

closed 2020-08-08 by svpernova09