PiPHP/GPIO #26: Under New Management :D

svpernova09, 2020-08-07:
This morning @AndrewCarterUK put out a call for a new maintainer: https://twitter.com/AndrewCarterUK/status/1291697455663325184

@Sam-Burns and I offered to step in and take over the project and I'm excited to say hello and share some initial plans with everyone.

A general  outline of my plans for the GPIO project:

* review/merge/decline the PRs as needed.
* Figure out what PHP versions we should support
* Swap to Github Actions because why not?! :D
* Start refactoring to support newer PHP versions

I'd love to know what OS our users are using on their Pis, here is a short straw poll: http://www.strawpoll.me/20734890

Raspberry Pi OS has PHP 7.0 -> 7.3 it seems. I think we start with 7.3 as a base target of compatibility and drop everything from there unless there is a very pressing need? If you have such a use case please comment!

If you have any questions for us please ask them here!

Alexander-Zierhut, 2021-01-08:
I am currently using PHP 8.0 and it works fine. What exactly are you trying to refactor? What do you think about adding abstractions for example: Buttons, Keypads, LCD Displays, etc. (I might be interested in creating a PR for that. )

svpernova09, 2021-01-08:
Anything to help modernize and improve readability I'd be all for.

Alexander-Zierhut, 2021-01-08:
Great! What do you think about making GPIO static?
So instead of:
```PHP
$gpio = new GPIO();
$pin = $gpio->getInputPin(2);
```

this
```PHP
$pin = GPIO::getInputPin(2);
```

This can also be implemented without breaking changes.

Alexander-Zierhut, 2021-01-08:
When you say `modernize`, are you facing any particular issues? I am just using it since two days and it seems pretty great to me. It's not even a big codebase. Very cool project indeed. I have been using C for these kinds of projects before.