PiPHP/GPIO #28: php version not declared

garak, 2020-08-08:
No php version is declared, indeed the whole "require" section is missing from composer.json (didn't look at code, so maybe some other ext could be missing too)

svpernova09, 2020-08-08:
We don't currently declare a PHP version, I'm brand new to the repo and am currently trying to sort out *which* versions of PHP we should support. Based on default Raspberry Pi OS IMHO we should be supporting 7.2, 7.3, and working to sort out how to get 7.4 installed and ensure working there.

I'm not sure what you were trying to do, but the readme instructions are correct in regards to running `composer require piphp/gpio` to install this into your application.

garak, 2020-08-08:
What I'm trying to do is improve this repository. Declaring supported php version is a de facto standard (apart from obvious advantages).
IMHO you should support nothing before php 7.2 (7.1 is too old and out of support) and 8.0.
So: `^7.2 || ^8.0`

svpernova09, 2020-08-08:
> 
> 
> What I'm trying to do is improve this repository. Declaring supported php version is a de facto standard (apart from obvious advantages).
> IMHO you should support nothing before php 7.2 (7.1 is too old and out of support) and 8.0.
> So: `^7.2 || ^8.0`

Ah I see. As it currently stands I believe i"m going to pick 7.3 to standardize on at the moment since 7.4 doesn't come default with Raspberry Pi OS.

closed 2020-08-08 by svpernova09

garak, 2020-08-08:
Not supporting current php version is not fair. Moreover, curren Ubuntu for Pi comes with php 7.4

svpernova09, 2020-08-08:
I'm fairly certain release 0.1.0 will work on PHP 7.4. 

Based on the 1 result in the poll on #26, the only respondent so far is using Raspberry Pi OS

[image]

I don't think it's too far off to get 7.4. Maybe even later today. There's just been several other things I wanted to get updated first.

garak, 2020-08-09:
Well, you can count me as a late voter for Ubuntu, taking your results to 50/50.
Anyway, you should release 0.1.0 as your *first* release, and targeting 1.0.0 as your stable (everything below 1 is de facto considered unstable)