PiPHP/GPIO #41: Installation guide insufficient

OZ1SEJ, 2020-09-28:
I installed via composer:

    $ php composer.phar require piphp/gpio
    Using version ^0.4.0 for piphp/gpio
    ./composer.json has been created
    Loading composer repositories with package information
    Updating dependencies (including require-dev)
    Package operations: 1 install, 0 updates, 0 removals
      - Installing piphp/gpio (0.4.0): Downloading (100%)         
    Writing lock file
    Generating autoload files

And then using this code in `/var/www/html/index.php`

    use PiPHP\GPIO\GPIO;
    use PiPHP\GPIO\Pin\PinInterface;
    
    // Create a GPIO object
    $gpio = new GPIO();
    
    // Retrieve pin 18 and configure it as an output pin
    $pin = $gpio->getOutputPin(4);
    
    // Set the value of the pin high (turn it on)
    $pin->setValue(PinInterface::VALUE_HIGH);

Results in

    Fatal error: Uncaught Error: Class 'PiPHP\GPIO\GPIO' not found in /var/www/html/index.php:13 Stack trace: #0 {main} thrown in /var/www/html/index.php on line 13

samwilson, 2021-01-06:
You need to include the Composer autoloader:

```php
require __DIR__.'/vendor/autoload.php';
```

paulisidore, 2022-03-29:
Hello, please add information that the apache2 user must be added to the gpio group if we want to deal with the web browser and that it is also necessary to restart raspiberry pi before using the library.

I was stuck on this point and finding the solution took a very long time (5 hours!!!). Thanks for the work and keep it up

OZ1SEJ, 2022-04-16:
This seems to solve the problem - closing.

closed 2022-04-16 by OZ1SEJ