PiPHP/GPIO #43: Permission denied for /sys/class/gpio/export

TinoArts, 2021-02-03:
OK, I really believe I have tried everything. I am constantly getting the permission fatal error:

`PHP Fatal error:  Uncaught RuntimeException: fopen(/sys/class/gpio/export): failed to open stream: Permission denied in /var/www/html/vendor/piphp/gpio/src/FileSystem/FileSystem.php:55`

I've tried every solution I was able to find online. I've tried adding www-data user to gpio group, which strangely resulted in Apache not starting at all (not sure why). I even tried dumb last resorts as setting permissions to 777 of whole /sys/class/gpio folder recursively, or adding unlimited access to www-data user in /etc/sudoers.

Still the same error. Any help?

AndrewCarterUK, 2021-02-03:
It sounds like you've probably tried/seen this, but it's the only relevant resource I'm aware of:

https://github.com/ronanguilloux/php-gpio#understanding-io-permissions

TinoArts, 2021-02-03:
Nevermind, this is completely my bad.

The key solution was to add www-data user to gpio group. I was thinking the Apache got broken by that, because by web app wouldn't run. Turned out it was just misimplementation of the code on my side which caused infinite loop (watch for PIN changes) before the page got rendered. Thank you.

closed 2021-02-03 by TinoArts