PiPHP/GPIO #45: fwrite(): Write of 2 bytes failed with errno=22

forenkema, 2024-12-07:
I have a PHP application which uses piphp. This worked fine on a Raspberry 4. However, after migrating to Raspberry 5, I get the exception: **fwrite(): Write of 2 bytes failed with errno=22:**

```
1. in /var/www/basic/vendor/piphp/gpio/src/FileSystem/FileSystem.phpat line 55
46474849505152535455565758596061626364        
$this->exceptionIfFalse($bytesWritten, $extra);
 
        return $bytesWritten;
    }
 
    private function exceptionIfFalse($result, $extra = '')
    {
        if (false === $result) {
            $errorDetails = error_get_last();
            throw new RuntimeException($errorDetails['message'].' '.$extra);
        }
    }
 
    /**
     * {@inheritdoc}
     */
    public function exists($path)
    {
        return file_exists($path);
2. in /var/www/basic/vendor/piphp/gpio/src/FileSystem/FileSystem.php at line 46– PiPHP\GPIO\FileSystem\FileSystem::exceptionIfFalse()
40414243444546474849505152    
{
        $stream = $this->open($path, 'w');
 
        $bytesWritten = @fwrite($stream, $buffer);
        fclose($stream);
        $extra = $path.' '.$buffer;
        $this->exceptionIfFalse($bytesWritten, $extra);
 
        return $bytesWritten;
    }
 
    private function exceptionIfFalse($result, $extra = '')
    {
3. in /var/www/basic/vendor/piphp/gpio/src/Pin/Pin.php at line 166– PiPHP\GPIO\FileSystem\FileSystem::putContents()
160161162163164165166167168     
     *
     * @param string $file The file to write to
     */
    private function writePinNumberToFile($file)
    {
        //echo "file is $file";
        $this->fileSystem->putContents($file, $this->getNumber());
    }
}
4. in /var/www/basic/vendor/piphp/gpio/src/Pin/Pin.php at line 62– PiPHP\GPIO\Pin\Pin::writePinNumberToFile()
56575859606162636465666768     */
    public function export()
    {
        if (!$this->isExported()) {
            $this->exported = true;
 
            $this->writePinNumberToFile($this->getFile(self::GPIO_FILE_EXPORT));
 
            // After export, we need to wait some time for kernel to report changes.
            usleep(200 * 1000);
        }
    }
```
 
Debugging shows that the $file = /sys/class/gpio/export and the $buffer = 16. Both seem totally fine to me....

My user (www-data) is in the gpio group (and in the root group). So I believe that file-permissions are not the issue.

Looking around on the internet, it appears that fwrite() errno 22 is caused by writing too much data to a file. That does not seem applicable here, as /sys/class/gpio/export seems to be an emulated file (Is that right? I am not sure).

braisdesousa, 2025-04-07:
Hi @forenkema ! Same problem here,  I did upgrade my raspberry, and that seems to be the problem. I'm installing right now and older versiones of the kernel. 

Maybe I'm mistaken but after hours of searching and little help from ChatGPT could be related with the Legacy gpio_sysfs  , and uses the libgpiod. 

But until I can install everything on a new card. It's only an asumption.

forenkema, 2025-04-08:
Hi Braisdesousa, ... if I am not mistaken I resolved this by switching over to pinctrl shell command which is part of the Raspberry 5 standard functionality. pinctrl lt's you address the gpio pins. Here is some of my PHP code.

private function pinctrl_set ($b_connect_nc)
	{
		$st_warning = $b_connect_nc ? 'b_connect_nc is true' : 'b_connect_nc is false';
		Yii::warning ($st_warning);
		if (Self::BooleanReversePolarity($this->pin)) {
			Yii::warning ('.... reversing polarity...');
			$b_connect_nc = !$b_connect_nc;
		}
		$st_warning = $b_connect_nc ? 'b_connect_nc is true' : 'b_connect_nc is false';
		Yii::warning ($st_warning);
		// assemble the shell 
		$st_command = $b_connect_nc ? 'dl' : 'dh';
		$st_command = "pinctrl set $this->pin op $st_command 2>&1 ";
		$response = exec( $st_command );
		Yii::warning($st_command); // in case something went wrong...
		Yii::warning("response is $response"); // in case something went wrong...
	}	
	
	public function pinctrl_level_exec ()
	{
		// make sure the gpio pin is set to 'op':		
		exec ("pinctrl set  $this->pin  op");
		
		$st_command = "pinctrl lev $this->pin 2>&1 ";
		return exec( $st_command );
	}
	
	private function pinctrl_level ()
	{
		$response = $this->pinctrl_level_exec();
		$b_pin_is_high = ('1' === $response);
		return $b_pin_is_high xor Self::BooleanReversePolarity($this->pin);
		switch ($response) 
		{
			case '0':
				// pin is low: this SHOULD be NC connected (unless polarity is reversed)
				return true xor Self::BooleanReversePolarity($this->pin);// We queried is_NC_ConnectedToCommon  ? ($b_testing was true)
			case '1':
				// pin is low: this SHOULD be NO connected (unless polarity is reversed)
				return false xor Self::BooleanReversePolarity($this->pin); // We queried is_NC_ConnectedToCommon  ? ($b_testing was true)
			default:
				Yii::warning($st_command); // in case something went wrong...
				Yii::warning("response is $response"); // in case something went wrong...
				return 'default';
		}

	}

braisdesousa, 2025-04-09:
Thanks @forenkema  I will try it. Right , for the purpose of my program I can for now keep my raspberry "old". But definitely I'll try your idea. 

Thank you.